fpgasystems/Coyote #9: There are some problems with the DDR setup.

qiuhaojie, 2022-09-27:
I want to use the "striping" example and execute the following instructions

cmake -DFDEV_NAME=vcu118 -DEXAMPLE="striping" ..

But there was an error. What caused it?
[image]

qiuhaojie, 2022-09-29:
> The MIG generates a different interface for vcu118 in comparison to Alveo cards. Seems to be related to ECC (whether it's enabled or not). I pushed a quick fix, it should build now.

Ok, I'll try again. Thank you

qiuhaojie, 2022-09-30:
Oh sorry, the last problem solved, a new problem appears, it seems that the port is not bound
[image]

d-kor, 2022-10-13:
The name of the port in the constraint file was wrong reset_nb instead of sys_reset_nb. Should be fixed now, but haven't given it a run.

qiuhaojie, 2022-10-18:
Thank you very much. Your work really helped me a lot, and I also learned a lot from it. I still failed to understand some of the operands in the process of using Coyote cProc::invoke, such as the difference between READ, WRITE, TRANSFER, OFFLOAD and SYNC. I see that TRANSFER is both READ and WRITE, I don't know when to use READ? When to use WRITE? When can I TRANSFER it? When to use OFFLOAD and SYNC? What's the difference between them? TRANSFER is sending data to the FPGA and reading it back, so why not READ and then WRITE? I did not find a similar explanation in your paper. I would be very grateful if you could answer my doubts.

d-kor, 2022-10-18:
All of the operations within invoke are data movement operations and they are all FPGA centric (the READ will read from the host to the user logic, the opposite applies for the WRITES). The TRANSFER is the simple combination of these two operations (think of it as a loopback), the advantage is that you only need to issue one command to initiate the TRANSFER, unlike READ+WRITE which would require two. This reduces the latency as there is no additional command over the PCIe. As we are using AVX there is plenty of space to provide the information within a single command. The OFFLOAD and SYNC are used to move data to/from FPGA side memory (DRAM/HBM). These are explicit commands for this movement, otherwise you can also rely on page faults to handle the OFFLOADs for you, for instance. Hope it clears it a bit.

qiuhaojie, 2022-10-18:
I see, thank you very much for your answer, your answer is very clear, let me out of the difficult situation, thank you very much indeed, thank you again.

qiuhaojie, 2022-10-18:
Oh, sorry, I have another problem. In the instance gbm_dtrees you gave, there is a register 
**cproc.setCSR(0x1, 0); // ap_start** 
,then transfers the data to FPGA by invoke to run. Also, in the function testcount in pr_scheduling, 
**cthread->setCSR(0x1, 0); // Start kernel** 
,then transfers data via invoke. However, the rotation function in pr_scheduling runs directly through invoke rather than through register setCSR. So I want to ask what is the difference between kernel startup through register and invoke startup and execution? For example, if I start the program after setCSR, but haven't transferred the data via invoke yet, what will happen? Shouldn't you first transfer the data to the FPGA via invoke and then start the execution by setting the register setCSR? Sorry to bother you.

d-kor, 2022-10-18:
The setCSR function doesn't handle data transfers. It propagates the start signal to the kernel to prepare for the new kernel invocation in this case. In general, what this function exactly does is kernel dependent. The CSR(Control/Status Registers) interface is a way to pass some form of information to/from your kernel. This is not tied to the invoke operations which are kernel independent and which control the data flow. If you have further questions about the framework operation, send us an email. Please keep this area for the actual issues with the code.

qiuhaojie, 2022-10-18:
Okay, thank you very much.

closed 2022-11-08 by d-kor