fpgasystems/Coyote #14: A bug occurred when canceling AVX

qiuhaojie, 2022-10-24:
After I create the project by executing the following instructions
**cmake -DFDEV_NAME=u50 -DEXAMPLE=gbm_dtrees -DEN_AVX=0 ..
make shell
make compile**
The following error occurred at compile time:
**ERROR: [Synth 8-524] part-select [95:64] out of range of prefix 'axi_rdata_bram' [/home/me/Coyote/hw/hdl/slave/cnfg_slave.sv:685]
ERROR: [Synth 8-6156] failed synthesizing module 'cnfg_slave' [/home/me/Coyote/hw/hdl/slave/cnfg_slave.sv:33]
ERROR: [Synth 8-6156] failed synthesizing module 'tlb_region_top' [/home/me/Coyote/hw/hdl/mmu/tlb_region_top.sv:42]
ERROR: [Synth 8-196] conditional expression could not be resolved to a constant [/home/me/Coyote/hw/hdl/mmu/tlb_top.sv:158]
ERROR: [Synth 8-6156] failed synthesizing module 'tlb_top' [/home/me/Coyote/hw/hdl/mmu/tlb_top.sv:41]
ERROR: [Synth 8-6156] failed synthesizing module 'design_dynamic_wrapper' [/home/me/Coyote/hw/build/lynx/hdl/wrappers/common/dynamic_wrapper.sv:8]
ERROR: [Synth 8-6156] failed synthesizing module 'top' [/home/me/Coyote/hw/build/lynx/hdl/wrappers/common/top_u50.sv:11]**

d-kor, 2022-10-24:
I updated the range of axi_rdata_bram,  could you give it another run?

qiuhaojie, 2022-10-26:
After our modification, the HW part was compiled successfully, but it will be stuck in this position in the figure when gbm_dtress is executed in u50. Is the option of AVX necessary?
[image]

d-kor, 2022-11-08:
fixed

closed 2022-11-08 by d-kor