fpgasystems/Coyote #22: Segmentation fault

qiuhaojie, 2022-11-19:
After I finished compiling the hw section and burning the bit flow through the jtag interface into vcu118, I executed
insmod coyote_drv.ko
A segment error has occurred, as shown below
[image]
And the printed kernel information is as follows:
[image]
The read configuration information is all 1. I do not know the cause

d-kor, 2022-11-21:
Did your rescan the PCIe bus?

qiuhaojie, 2022-11-21:
Yes, restart the host after burning the bitstream. After restart, check the device before loading the driver and input instructions
lspci -vv -d 10EE:
However, the PCIe device is not found.
And then execute
echo 1 > /sys/bus/pci/rescan
Also run lspci -vv -d 10EE:
The PCIe device is still not found.

qiuhaojie, 2022-11-21:
Yes, restart the host after burning the bitstream. After restart, check the device before loading the driver and input instructions
lspci -vv -d 10EE:
[image]
However, insmod coyote_drv.ko will still appear Segmentation fault.

qiuhaojie, 2022-11-21:
All the information is as follows:
[image]

d-kor, 2022-11-24:
What is the benchmark you are running and what is the config?

qiuhaojie, 2022-11-24:
The instructions I entered were:

cmake -DFDEV_NAME=vcu118 -DEXAMPLE=gbm_dtrees ..

make shell compile

Then load the bitstream to vcu118, restart the host, and load the driver to produce this error

aaqdas, 2024-09-12:
I had a similar issue. I was trying to run the `insmod` in a virtual machine. When I tried running this on the actual machine, it worked. See if that helps

closed 2024-11-14 by bo3z