fpgasystems/Coyote #31: Runtime error in perf_host

surangamh, 2023-02-13:
I built the perf_host design for U280. I could generate the bitstream, program the alveo, and load the driver. However, I got the following error when I tried to run the sw executable.

Number of regions: 3
Hugepages: 0
Mapped pages: 1
Number of allocated pages: 8
Number of repetitions: 100
Starting transfer size: 128
Ending transfer size: 32768
terminate called after throwing an instance of 'std::runtime_error'
  what():  cProcess could not be obtained, vfid: 0
Aborted (core dumped)

Any thoughts on how to debug?
Thanks!

yluo39github, 2023-02-22:
Same error, 
I can find fpga0, 1, 2 in /dev/ folder. However, fd output is still -1.

yluo39github, 2023-02-22:
lol, just $ sudo ./main

surangamh, 2023-03-18:
@yluo39github Thanks. Actually, the issue was that I didn't warm boot the machine after programming the bitstream.

closed 2023-03-18 by surangamh