fpgasystems/Coyote #47: Kernel BUG on probe

mar-ven, 2023-11-17:
I am experiencing the following kernel BUG at Coyote/driver/fpga_dev.c:108
I am currently using an Alveo U280. Linux Kernel: 6.5

My steps:

- I made sure that coyote driver was not loaded
- I used Vivado Hardware Manager 2022.2 to load the bitstream
- I ran sw/util/hot_reset 
- lspci | grep Xilinx shows that the board is correctly identified
- I launch sudo insmod coyote_drv.ko

This last command hangs, because of the kernel bug.

I experienced this error previously, with the same setup apart from the Linux kernel, which was 5.15

mar-ven, 2023-11-17:
Some time before this error happened, I experienced another error: AMD-Vi IO_PAGE_FAULT, while I was launching the test.cpp file with options `-f -r`. This error happened before any test functions were actually run

closed 2024-10-22 by bo3z

HCC2620, 2025-05-09:
> I am experiencing a kernel BUG at Coyote/driver/fpga_dev.c:108 (accl_integration branch) I am currently using an Alveo U280. Linux Kernel: 6.5
> 
> My steps:
> 
> * I made sure that coyote driver was not loaded
> * I used Vivado Hardware Manager 2022.2 to load the bitstream
> * I ran sw/util/hot_reset
> * lspci | grep Xilinx shows that the board is correctly identified
> * I launch sudo insmod coyote_drv.ko
> 
> This last command hangs, because of the kernel bug.
> 
> I also experienced this error previously, with the same setup apart from the Linux kernel, which was 5.15

Hello！
I also experienced this BUG.Did you find the reason and solution?
Thanks

bo3z, 2025-05-09:
Hi @HCC2620, which branch of Coyote are you using?

HCC2620, 2025-05-09:
Hi @bo3z ,I am using this branch:https://github.com/fpgasystems/Coyote/tree/ef4853fc4eefc2768213179abe1b9278834bd2d2

bo3z, 2025-05-09:
Could you check out this branch: https://github.com/fpgasystems/Coyote/tree/tutorial. This is a more up-to-date branch with Coyote v2. It's still experimental and not part of main, but more stable then the ACCL branch; we are hoping to merge it soon. It also includes a number of examples and tutorials on getting started.

bo3z, 2025-05-11:
No, Coyote doesn't support for Alveo U50. The target branch mainly supports U55C, U250 and some limited support for the U280.