fpgasystems/Coyote #52: OS hanging when fpga_tlb_miss_isr():(irq=107) page fault ISR

crizy, 2024-04-11:
When a missing page prompt appears and the missing page exception is handled, the operating system hangs. 
dmesg information is as follows，
         fpga_tlb_miss_isr():(irq=107) page fault ISR
         fpga_tlb_miss_isr():page fault, vaddr 7ffbc5402000, length 40, cpid 0
         tlb_get_user_pages():pid found = 2823
         tlb_get_user_pages():allocated 8 bytes for page pointer array for 1 pages @0x000000001bd3b688, passed size 64.
         tlb_get_user_pages():pages=0x000000001bd3b688
         tlb_get_user_pages():first = 7ffbc5402, last = 7ffbc5402
         tlb_get_user_pages():get_user_pages_remote(7ffbc5402000, n_pages = 1, page start = 7a19000000, hugepages = 0)
         tlb_get_user_pages():could not get all user pages, -14
        fpga_tlb_miss_isr():pages could not be obtained

When I use a large memory page (HUGE_2M), the dmesg information is also similar to the above dmesg information when there is a page missing prompt, and the dmesg information shows hugepages = 0 (and the large memory page is not started)

d-kor, 2024-05-03:
Could you try with the latest updates?

crizy, 2024-05-04:
> Could you try with the latest updates?
@d-kor，
          Thank you very much for your response，
          I have found that the latest version has undergone significant changes compared to previous versions, including overall architecture (configurable), hardware MMU, and software shared memory methods, among others

closed 2024-05-30 by crizy