fpgasystems/Coyote #63: QDMA Support

HongshiTan, 2024-05-02:
Hello, 

This is Hongshi from NUS HACC.  May I know if there is any schedule for supporting QDMA? 

If not, I would like to contribute to QDMA-related features for Coyote. Since the current project is quite large, please let me know if there are any details, like the milestones, etc., I need to follow.

d-kor, 2024-05-03:
Hi Hongshi, 

We don't have any plans to do QDMA support on our side. We utilize either the XDMA (it has a much higher reliability than QDMA) or our own internal DMA for our custom hardware platform (Enzian). 

We did have ideas of plugging in QDMA at some points in time, but never really had the manpower to try it out. It would be great if you are be willing to explore the idea and we could provide some support from our side if needed.

We can start by creating a dev branch dedicated to qdma where you can open up pull requests.

HongshiTan, 2024-05-03:
Sure! I will try it, thanks for the information about QDMA's reliability. I will update you if there is any progress from my side.

closed 2024-10-22 by bo3z