fpgasystems/Coyote #64: bitgen error

zjb2023, 2024-05-28:
[image]

[image]

the vFPGA_top.sv 
intsanted some modules can't be find can anyone help to find

rdma_base_slv inst_rdma_base_slv (
    .aclk(aclk),
    .aresetn(aresetn),

    .axi_ctrl(axi_ctrl),

    .mux_ctid(mux_ctid)
);

mux_host_card_rd_rdma inst_mux_send (
    .aclk(aclk),
    .aresetn(aresetn),

    .mux_ctid(mux_ctid),
    .s_rq(rq_rd),
    .m_sq(sq_rd),

    .s_axis_host(axis_host_recv[0]),
    .s_axis_card(axis_card_recv[0]),
    .m_axis(/*axis_rdma_send[0]*/)//
);  

mux_host_card_wr_rdma inst_mux_recv (
    .aclk(aclk),
    .aresetn(aresetn),

    .mux_ctid(mux_ctid),
    .s_rq(rq_wr),
    .m_sq(sq_wr),

    .s_axis(/*axis_rdma_recv[0]*/),//
    .m_axis_host(axis_host_send[0]),
    .m_axis_card(axis_card_send[0])
);

JonasDann, 2024-09-16:
On our end building Coyote with RDMA works. Please give a detailed explanation what exactly you are trying to do. Otherwise, we can't help you.

bo3z, 2024-11-27:
Closing due to inactivity. Please re-open at a later stage if still applicable.

closed 2024-11-27 by bo3z