fpgasystems/Coyote #75: RDMA Server Exits Abruptly

aaqdas, 2024-09-13:
I've synthesized the design and ran the hardware on AU-250. It works properly and shows me that I have **RDMA Enabled 1** in the `dmesg` logs and provides me with a MAC and IP.

When I try and run the `rdma_server` application, the application gives the following output and exits abruptly.

I'm assuming that the `device` and `vfid` is the number thats listed in `ls -la /dev/fpga_{d}_v{i}`. I use the following command to run the `rdma_server` 

```
./bin/test -d 0 -i 0
```
which gives the output
```
Forking...

```

Has anyone face anything similar, or is there a bug that needs to be addressed?

JonasDann, 2024-09-16:
It is normal that the server forks and runs in the background. Can you not see it running in htop afterwards?

aaqdas, 2024-09-17:
It shows up in `htop`. However, it doesn't generate any logs. I'm assuming it is supposed to generate a log, something similar to what is shown [https://github.com/fpgasystems/Coyote/issues/33](url).

I see that you've updated the documentation for RDMA, which says that I'd need to have FPGAs on both ends. 

**I am trying to ping between a Mellanox Connect-X6 and RoCE Implementation of Coyote. Is that supported?**

aaqdas, 2024-09-17:
I have also set up the alternative branch `coyote-v1` on my systems. In that branch, the program is killed during the execution of the function `ibvQpMap::addQpair` and `SIGKILL` is returned. I've also been debugging it, and it appears to be the problem somewhere at `ibvQpMap.cpp:30`.

aaqdas, 2024-09-30:
ibvQpMAp.cpp:30 problem was due to HUGE_PAGES. It can be solved by setting the huge pages with this command.

```bash
echo 1024 > /proc/sys/vm/nr_hugepages #The Number of Huge Pages Go In This File
```

aaqdas, 2024-10-15:
Hi,

Thank you for your support. I was able to generate the traffic and receive it on the other end.

closed 2024-10-15 by aaqdas