fpgasystems/Coyote #78: Each completion queue contains 2-cycle burst valid signal

zhenhaohe, 2024-09-29:
Hi all,

Recently I found a potential bug with the completion queue of Coyote. Each compeletion of the request comes with a 2-cycle burst valid signal instead of a 1-cycle valid.

maximilianheer, 2024-10-08:
Hi @zhenhaohe or @quetric, can we maybe get a bit more context on this issue? Which scenario, which busses are we talking about? Do you maybe even have a screenshot of waveforms or something like this?

quetric, 2024-10-09:
This bug is observed on the `cq_wr` bus. I don't have a screenshot of waveforms, but the context is that after a write request is issued on `sq_wr`, we are observing `cq_wr` for a completion. The bug is that instead of TVALID staying high for 1 cycle to signal a completion, it stays high for 2 consecutive cycles.

maximilianheer, 2024-10-09:
Hi @quetric, thanks a lot for the clarification. We'll be looking into it.

bo3z, 2025-02-20:
Fixed in this commit: https://github.com/fpgasystems/Coyote/commit/b8921b7e189bf8985fec2b40b502adac8f67970b. 

PR of the branch to come soon. Closing for now.

closed 2025-02-20 by bo3z