fpgasystems/Coyote #103: rtl codes signal problems

crizy, 2025-01-02:
the path :  hw/hdl/network/rdma/
the file   :  rdma_meta_tx_arbiter.sv

the rtl code at line 163 of rdma_meta_tx_arbiter.sv file,
   
     assign user_seq_in.data = {vfid, request_snk[vfid].req_1.len};

the request_snk[vfid]  signal does not exist in rdma_meta_tx_arbiter.sv file, what is in the file is just req_snk[vfid],

so line 163 should be :

     assign user_seq_in.data = {vfid, req_snk[vfid].req_1.len};

crizy, 2025-01-14:
the path :  hw\services\network\hls\hash_table\
  the file   : generate_random_table.cpp  
 
  the hls code at line 61 of generate_random_table.cpp file,
         uint32_t maxKeyValue = pow(keyLength, 2) - 1;

  should be :
         uint32_t maxKeyValue = pow(2, keyLength) - 1;

bo3z, 2025-01-14:
Hi @crizy,

Thank you for noticing both of these. Can you please explain a bit what work-loads (protocols, numbers of instances, connections etc.) that you are using. These are obvious syntax errors that we will fix, but for some reason we don't encounter them in our applications. So we would like to understand if there are some other subtle issues for work-loads we may not be using.

Thanks!

crizy, 2025-01-22:
Hi @bo3z ,

     the first rtl syntax errors (rdma_meta_tx_arbiter.sv) was reported when I synthesized the rdma_perf example using U250 FPGA device
     the second （generate_random_table.cpp）was discovered when I was reviewing the HLS code

crizy, 2025-01-23:
Hi @bo3z ，
      
     Because the path hw/checkpoints does not have a static u250_satic.dcp file, I only combined the creation of a static project and rdma_shell project together, and added the following static_rdma example in the file examples_hw/CMakeLists.txt

if(EXAMPLE STREQUAL "static_rdma")
    message("** static RDMA perf test")
    set(BUILD_STATIC 1)
    set(BUILD_SHELL 0)
    set(BUILD_OPT 1)
    set(N_REGIONS 4)
    set(EN_STRM 1)
    set(EN_MEM 1)
    set(EN_RDMA 1)
    set(N_STRM_AXI 2)

    validation_checks_hw()

    load_apps (
        VFPGA_C0_0 "apps/rdma_perf"
        VFPGA_C0_1 "apps/rdma_perf"
        VFPGA_C0_2 "apps/rdma_perf"
        VFPGA_C0_3 "apps/rdma_perf"
    )

    create_hw()
endif()

I executed this static_rdma example to create a project with U250 fpga and generate a bit file, but the shell project reported a synthesis failure for the rdma_meta_tx_arbiter.sv file during synthesis

crizy, 2025-01-23:
hi @bo3z ,

    Because the path hw/checkpoints does not have a static u250_datic.dcp file, 
I combined the creation of a static project and rdma_sthell project together, 
and added the following static_rdma example in the file examples_hw/CMakeLists.txt

    if(EXAMPLE STREQUAL "static_rdma")
        message("** static RDMA perf test")
        set(BUILD_STATIC 1)
        set(BUILD_SHELL 0)
        set(BUILD_OPT 1)
        set(N_REGIONS 4)
        set(EN_STRM 1)
        set(EN_MEM 1)
        set(EN_RDMA 1)
        set(N_STRM_AXI 2)
    
        validation_checks_hw()
    
        load_apps (
            VFPGA_C0_0 "apps/rdma_perf"
            VFPGA_C0_1 "apps/rdma_perf"
            VFPGA_C0_2 "apps/rdma_perf"
            VFPGA_C0_3 "apps/rdma_perf"
        )
    
        create_hw()
    endif()

    I executed this static_rdma example to create a project with u250 fpga and generate a bit file, but the 
shell project reported a synthesis failure for the rdma_meta_tx_arbiter file during synthesis