fpgasystems/Coyote #136: Iperf TCP/UDP using Coyote

adityapeer7, 2025-07-29:
We have an Alveo U55C card, and our current setup is that the wires are connected in the loopback mode. We want the network data to go from the host to the Alveo board, and then the network data is sent through one port and received back at another port, then it should be sent back to the host. 

Is there any Iperf test example to test the bandwidth? We want to test for TCP/UDP. We are aware of the RDMA stack.

bo3z, 2025-08-11:
Hi @adityapeer7, unfortunately we don't (currently) have any guides on running TCP/IP on Coyote. We are looking into bringing it back in Coyote v2, but it's unclear how long / when this development will take place. 

Reading the set-up you have, it should also be possible to use RDMA for your test?

adityapeer7, 2025-08-11:
Hi @bo3z I tried the RDMA test, Initially, I generated and loaded the bitstream for the perf_RDMA example and inserted the coyote_driver with giving IP and MAC address, but not able to see the interface for the QSFP ports to be up when do "ifconfig" command, is something I am missing here ? 
Currently I have inserted the IP and MAC address given in the directory : /sys/kernel/coyote_sysfs_0/ 

The second issue I faced currenlty I see two pci addresses when I ran this command (lspci -nnk -d 10ee:) but when I do pci_hot_plugin.sh script, 9d:00.0 remains, but 9d:00.1 goes off, after reboot only it comes back but binds with existing xocl and xclmgmt drivers. I want to give QSFP ports interfaces to perform the loopback test. 

Can you please guide me to resolve these two issues.

adityapeer7, 2025-08-11:
Hi @bo3z, the second issue, initially, when I run the command : 

- sudo lspci -nnk -d 10ee:

It gave the  following output :

9d:00.0 Processing accelerators [1200]: Xilinx Corporation Device [10ee:505c]
        Subsystem: Xilinx Corporation Device [10ee:000e]
        Kernel driver in use: xclmgmt
        Kernel modules: xclmgmt

9d:00.1 Processing accelerators [1200]: Xilinx Corporation Device [10ee:505d]
        Subsystem: Xilinx Corporation Device [10ee:000e]
        Kernel driver in use: xocl
        Kernel modules: xocl

But after running the pci_hot_plug script to insert the coyote driver, I am not able to see 9d:00.1 pci address in the terminal, only 9d:00.0 is seen. My only concern is that there will be any issue in running the loopback test, because earlier I was thinking of assigning different IP addresses to QSFP ports.

adityapeer7, 2025-08-14:
Hi @bo3z , according to my current unsertadning to assign IP and mac address , we have to pass arguments with the insertion of coyote driver. Can you please tell how we particularly assign IP address to the different QSFP ports, and how it's been done internally with the help of Coyote driver ?