fpgasystems/Coyote #139: Invalid memory write in new Coyote thread constructor

sven-weber, 2025-08-27:
**Description**
After switching to the new Coyote release (v.0.2.1), our software experienced a segfault at the end of the program runtime. After some investigation, we found that this segfault only happens when the Coyote cThread is initialized. Therefore, we ran our program with Valgrind and found that there are out of bounds writes and reads that happen in the new cThread constructor. Those writes overwrite other memory of our software, which causes the segfaults at program exit. When going back to a previous Coyote version, this issue does not happen.

Attached is the corresponding valgrind run, which can be imported, for example in CLion. The issue seems to stem from cTread line 108:

[image]

**Expected behavior**
Do not perform out of bounds writes and reads.

**Screenshots & Logs**

[valgrind.xml](https://github.com/user-attachments/files/22010251/valgrind.xml)

**Environment:**
 - OS (Ubuntu & Linux version): Ubuntu 22.04.5 LTS - Kernel 6.8.0-65-generic (HACC machine 09)
 - CMake version: 3.31.0-rc3
 - gcc version: 11.4.0
 - Vivado / Vitis version: 2024.2
 - Tested on hardware? No - pure software-side issue
 - If tested on hardware, what FPGA (e.g. Alveo U55C, Alveo U200 etc.):

**Additional information**

As the software is not publicly accessible, I am happy to share steps on how to re-produce this issue privately.

bo3z, 2026-09-24:
**Root cause:** the `cThread` object's binary layout depends on whether `EN_AVX` is defined, and the Valgrind trace shows the object being allocated in one binary and constructed in another — if those two disagree on `EN_AVX`, the allocation ends up smaller than the constructor needs, and the constructor writes past the end of it.

**The layout dependency:**

```cpp
// sw/include/coyote/cThread.hpp:100-102
#ifdef EN_AVX
volatile __m256i *cnfg_reg_avx = { 0 };
#endif
```

`EN_AVX` defaults to `1` in Coyote's own CMake (`sw/CMakeLists.txt:38,156-158`), and when set, `cThread` gains this extra 8-byte pointer member. `libcoyoted.so.2.0.0` — the binary the trace shows `cThread::cThread`/`~cThread` actually executing in — was built with it on.

**What the trace shows:** the allocation and the construction happen in two different binaries.
- `std::allocate_shared<coyote::cThread>(...)` runs inside `libmaximus.so` — that's where `_Sp_counted_ptr_inplace`, `__shared_count`, and `allocate_shared` resolve in the stack.
- The constructor body that then runs against that block, `coyote::cThread::cThread(...)`, resolves in `libcoyoted.so.2.0.0`.

`std::allocate_shared<T>` sizes its allocation from `sizeof(T)` as computed wherever it's instantiated — i.e. in whichever TU calls it, using whatever view of `cThread.hpp` the compiler saw there. If `EN_AVX` wasn't defined in that TU, it computes `cThread` 8 bytes smaller than it actually is (no `cnfg_reg_avx`). The allocation is then sized for that smaller layout, but the constructor that runs against it is the one built into `libcoyoted.so`, which assumes the larger layout and writes to member offsets accordingly. Once it reaches the members declared after `cnfg_reg_avx` (`cnfg_reg`, `ctrl_reg`, ..., `vlock`, `lock_acquired`), it's writing 8 bytes further into the object than was allocated — landing past the end of the heap block. That's the invalid write at `cThread.cpp:108`, and the invalid read at `~cThread()` (`cThread.cpp:197`) is the destructor touching that same past-the-end region during teardown.

**Why it shows up as a segfault elsewhere, later:** the out-of-bounds write itself doesn't corrupt Coyote's own state — it corrupts whatever heap metadata or adjacent allocation happens to sit right after that block. That's why the crash surfaces at program exit rather than at construction time.

**Why this looks like a build/link-configuration issue rather than a logic bug in `cThread`:** Coyote exports a proper CMake package (`install(EXPORT CoyoteTargets ...)` → `CoyoteConfig.cmake`) specifically so `EN_AVX` (and `EN_ROCM`/`EN_CUDA`) propagate as `PUBLIC` compile definitions to anything linking against it via `find_package(Coyote)` + `target_link_libraries(app PRIVATE Coyote)`. When that path is used, `EN_AVX` can't disagree between `libcoyoted.so` and the consuming application — they'd always share the same macro state. For the mismatch to occur, something is bypassing that propagation, most commonly:
- Linking directly against the installed `.so` (raw path/`-l`) instead of the exported `Coyote` CMake target, so `INTERFACE_COMPILE_DEFINITIONS` never reach that compile.
- Re-packaging/vendoring the built library and headers (the paths under `/local/home/svweber/dest/...` suggest a custom install step) in a way that drops the CMake package config.
- Different parts of the same build (`libmaximus.so` vs. `maxbench`) setting `-DEN_AVX` independently instead of inheriting it from one shared dependency.

**Worth checking:** how `Coyote` is consumed in the downstream `CMakeLists.txt` — `find_package(Coyote)` + the exported target, or a manually specified library path/include dir — and whether `EN_AVX` is defined identically for every target that includes `cThread.hpp`, including whatever built `libcoyoted.so.2.0.0` itself.

Regardless of the exact cause on the downstream side, this class shouldn't be fragile to this kind of mismatch in the first place — I'll push a fix that makes `cThread`'s layout independent of `EN_AVX`.

bo3z, 2026-09-24:
Fixed by PR #252. @JonasDann can you please review and confirm the fix addresses the issue. Once the PR is merged, we can close this issue.

closed 2026-10-05 by JonasDann